adafruit/Bluefruit_LE_Connect_v2 #19: iPhone differs from iPad

adamstener, 2017-03-03:
On the iPhone, if you filter out the devices based on signal strength, anything not matching the filter is removed from the list. If you use the same parameters on the iPad, it seems like anything advertised with terrible signal strength still makes the list. I would assume this has to do with an RSSI coming through that's a false positive value, and somehow this is managed on the phone differently than the iPad.

adamstener, 2017-03-03:
On the iPhone, I can dial down the signal strength window, that allows me to see how close I have to get to a device in order to pass that threshold.  On the iPad, there are too many things in the list to have this same effect, so I'm unable to perform the same tests, which would be extremely helpful if I could.

antonio-openroad, 2017-03-09:
Thanks for reporting it. 
It looks that it is related to how undefined RSSI values are handled

adamstener, 2017-03-09:
In my BLE implementation, I sometimes get arbitrarily large values, like 128, which need to be filtered out.  Was kind of thinking it could be this.

adamstener, 2017-03-13:
The latest commit has solved the issue, thanks much.

commit a321ad2d (2017-03-09): - DFU: better disconnection management. No longer reports the device is disconnected if the update didn’t actually disconnect it
- Scanning: better handling for RSSI when the operating system reports an undefined RSSI value fixes #19
closed 2017-05-24 by antonio-openroad