adafruit/Bluefruit_LE_Connect_v2 #20: Unclear if UART terminal is compatible with bluefruit 52832

rkertesz, 2017-04-13:
Is the UART terminal compatible with the new adafruit 52832 feather board? I ask because this post cites the iOS app but it also says you should custom build your own UART communication and that the api for the 52832 is different from the 51822 model (no 32u4 chip).

rkertesz, 2017-04-13:
Adafruit sales post. https://www.adafruit.com/product/3406

rkertesz, 2017-04-13:
Link to post on building your own UART comm service. https://learn.adafruit.com/bluefruit-nrf52-feather-learning-guide/bleuart

microbuilder, 2017-04-13:
Please post support request in the support forums, thanks. The support staff doesn't monitor Github issues, nor do they get the email alerts. But ... there is a 'bleuart' demo that works with the bluetooth UART service, and this demo connect the HW UART to the BLE UART, which should give you 90% of what you need.

closed 2017-04-13 by microbuilder

rkertesz, 2017-04-13:
It's not so much a support request as it is a question. I am willing to help here but wanted to know the status. In any event, I will follow your recommendation and go to support or I'll just dig through the code and answer the question myself. Thank you.

microbuilder, 2017-04-13:
There are just more eyeballs in the forums, is all. :) I think the bleuart example should do what you want, though ... it shows you how to use BLE UART and HW UART together, and feeds data back and forth between the two. This will behave identically on the iOS or Android versions of Bluefruit LE Connect.

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/Peripheral/bleuart/bleuart.ino

microbuilder, 2017-04-13:
Contributions to the app are of course always welcome!

rkertesz, 2017-04-13:
Oh I see. I misunderstood. Thank you

microbuilder, 2017-04-13:
This code from the bleuart demo should give you what you want. It scan HW UART for data, and if any is found sends it to BLE UART. If any data is found on BLE UART, it sends it out to HW UART, making a UART bridge over BLE. Simply open the 'UART' app in Bluefruit LE Connect and you will see the BLE side of any transactions, no additional code required on your end:

```
void loop()
{
  // Blinky!
  if (blinkyms+BLINKY_MS < millis()) {
    blinkyms = millis();
    digitalToggle(STATUS_LED);
  }

  // forward from Serial to BLEUART
  if (Serial.available())
  {
    // delay to get enough inputs, since we have limited number of transmission buffer
    delay(2);

    uint8_t buf[64];
    int count = Serial.readBytes(buf, sizeof(buf));
    bleuart.write( buf, count );
  }

  // forward from BLEUART to Serial
  if ( bleuart.available() )
  {
    uint8_t ch;
    ch = (uint8_t) bleuart.read();
    Serial.write(ch);
  }
}
```