adafruit/Bluefruit_LE_Connect_v2 #23: Swift 3.0?

nanohaddad, 2017-06-01:
Just wondering if you guys intend on updating the project or if its retired, are there any other examples on how to connect to bluefruit through iOS that are currently functioning? thanks.

antonio-openroad, 2017-06-01:
Please check the swift3 branch
The iOS app is already updated to swift3. It will be merged in master when the macOS app is also ported.

nanohaddad, 2017-06-01:
Thanks so much! I didnt think to check the other branches - sorry.

I'm trying to figure out how to query Bluefruit for battery life / communicate with it. Any guide on how to accomplish that? I haven't been able to get a response after connecting to it through the iOS app

microbuilder, 2017-06-02:
Sorry, we really can't offer any support on this code, it's published as is as a reference, but given the limited resources we have it really isn't possible to support Swift or any mobile development. There are far larger and more focused forums for that online. :(

closed 2017-06-02 by microbuilder