adafruit/Bluefruit_LE_Connect_v2 #31: Cannot build IOS project onto my phone to test

xt280, 2017-11-27:
Check dependencies

Code Signing Error: No account for team "2X94RM7457". Add a new account in the Accounts preference pane or verify that your accounts have valid credentials.

Code Signing Error: No profiles for 'com.adafruit.bluefruitconnect.watchkitapp.watchkitextension' were found:  Xcode couldn't find any iOS App Development provisioning profiles matching 'com.adafruit.bluefruitconnect.watchkitapp.watchkitextension'.

Code Signing Error: Code signing is required for product type 'WatchKit Extension' in SDK 'watchOS 4.1'

microbuilder, 2017-11-27:
You need to use your own Apple developper account when building this app. This is a reference to the Adafruit account which you obviously don't have available locally.

closed 2017-11-27 by microbuilder

xt280, 2017-11-27:
Thanks I got it working :)