adafruit/Bluefruit_LE_Connect_v2 #36: what is init file?

viveksoni95, 2019-02-07:
HI, 
I am trying to update firmware OTA. The bluefruit LE app is asking for an init file with hex file. 

What is init file and how to generate in arduino IDE ?

Thank you

skudave, 2020-09-27:
As per [https://github.com/adafruit/Adafruit_BluefruitLE_Firmware](https://github.com/adafruit/Adafruit_BluefruitLE_Firmware) 

filename_init.dat - This file contains meta-data about the main firmware image in filename.hex, such as the required SoftDevice version and expected HW. This file is required when performing over-the-air (OTA) updates, which is a two file process.

latne1ec, 2021-05-28:
Any idea how to generate that init file?

elral, 2022-07-22:
I am building my firmware with PlatformIO. After compiling two files are generate, firmware.hex and firmware.zip. The zip file contains 3 files, firmware.bin, firmware.dat and manifest.json.
In the past (more than 1.5 years) I used the firmware.dat file as init file (as far as I remember). But now this seems not to work anymore. I also upgraded the bootloader for my NRF5280 board to the newest versio 0.7.0, but it didn't change anything. I can see that the bootloader gets activated, but the upload is aborted and the bootloader is started again in loop.

Are there any hints how to get a successfull OTA update of a custom firmware?