adafruit/Bluefruit_LE_Connect_v2 #37: How to change device name "Adafruit Bluefruit LE" to "My Device".

Ilesh, 2019-03-06:
Hello Team (@martymcguire , @tannewt , @brennen , @microbuilder , @jwcooper ),

I am working on **Adafruit Bluefruit LE UART Friend - Bluetooth Low Energy (BLE)**. I have successfully implemented code for the communication with the device and it works well. I want to change the name which appears in the device search list. It displays  **Adafruit Bluefruit LE**. I have tried the following way but not work for me.

```
# Read the current device name
AT+GAPDEVNAME
UART
OK
 
# Update the device name to 'BLEFriend'
AT+GAPDEVNAME=BLEFriend
OK
# Reset to take effect
ATZ
OK
```
Please let me know what's wrong with me. Is there any other way to do that?
Thanks for your support.

Thanks regards,
Ilesh

tannewt, 2019-03-08:
Please do not mention any extra people unless you've previously had a conversation with them about the issue elsewhere.

ladyada, 2019-03-08:
@Ilesh 
this isn't an issue for this library
this isn't the appropriate place to ask for tech support

closed 2019-03-08 by ladyada