adafruit/Bluefruit_LE_Connect_v2 #41: Sending more than 20 bytes via UART

EThornill, 2020-03-13:
Hi! I would love to have the ability to send UART messages longer than 20 bytes from the Bluefruit Connect app (iOS/macOS, version 3.6.1). Currently messages over 20 bytes do not appear to get sent to the peripheral. 

This doesn't appear to be an issue for the old Bluefruit LE Connect for Mac0S (version 2.2.2), which sends longer messages just fine.

antonio-openroad, 2020-03-19:
I tested on both iOS and macOS and I can send messages longer than 20 bytes successfully. 
Could it be a problem on the peripheral side?

EThornill, 2020-03-20:
Thanks Antonio. I've done some testing with several of the ItsyBitsy nRF52840 expresses I have using example UART code from the nRF52/BLE libraries.

I only seem to have the problem when using CircuitPython and the Bluefruit Connect app. Arduino works fine… as does CircuitPython together with the old MacOS Bluefruit LE Connect program. 

So maybe it's a CircuitPython issue?

MakerThornhill, 2020-04-09:
Did some further testing using 2 boards to send UART message >20 bytes to each other. They didn't have any problems, using Arduino or CircuitPython. As far as I can tell, the peripherals are working… but CircuitPython + the Bluefruit Connect app does not.

dalise, 2020-04-11:
Not sure if this helps, but it may relate to the protocol data length of LE being 27 bytes 
4 L2CAP bytes.
3 ATT bytes
20 attribute bytes (data)
For more info, checkout the 29 min mark of the Apple WWDC 2017 712 conference as it goes into detail of the packet structure. Just search for "What's new in Core Bluetooth 712"

MakerThornhill, 2020-04-11:
Right, but I think data > 20 bytes gets put in a buffer and sent in packets… which is what the BLE app + arduino library seem to do… and my microcontrollers when they're sending messages to each other? 

I'm not sure why the BLE app + CircuitPython seems to have problems… or if it's the BLE connect app or CircuitPython.

antonio-openroad, 2020-04-12:
A comment: iOS is not limited to transfer 20 bytes per write. 
The MTU (maximum transfer unit) is negotiated between the devices on connection.
For example when connecting to an Adafruit LE Friend the limit is 20 bytes, but when connecting to a Feather with the firmware for ImageTransfer the limit is 244 bytes 

So, the bytes on each packet depends on the MTU negociated

closed 2021-11-10 by antonio-openroad