adafruit/Bluefruit_LE_Connect_v2 #44: Uart publish to more than a single Adafruit IO feed

systembolaget, 2021-03-15:
The Android (and iOS) app can only publish to one Adafruit IO feed. How can one publish to two (or more) instead?

As recommended in the Adafruit IO forum, we are using Adafruit Feather 32u4 Bluefruit LE with **two** different sensors (decibels, PM 2.5) connected to it, but found out yesterday on Discord help-with-adafruit-io that one can only publish the values of **one** sensor [https://learn.adafruit.com/bluefruit-le-connect/mqtt](https://learn.adafruit.com/bluefruit-le-connect/mqtt)

Can the app be enhanced to allow publishing to a second feed? Otherwise we'd have to use Adafruit Fona, which is costly for citizen science projects, and everyone already owns an Android (or iOS) smartphone, so using Bluefruit LE connect would be much better.

ladyada, 2021-03-15:
at this time, its not possible!

systembolaget, 2021-03-15:
Thanks, too bad : (

Hopefully the app will allow more feeds eventually, would be ideal for many citizen science projects.

closed 2021-03-15 by ladyada