adafruit/Bluefruit_LE_Connect_v2 #48: info file issue

Poojamok, 2022-03-15:
App is running on simulator but not on real device and also not archieve. whenever i am trying to running on real device it give me following error, 

** #The value of CFBundleShortVersionString in your WatchKit app's Info.plist (3.6.2) does not match the value in your companion app's Info.plist (3.7.0). These values are required to match.**

could you please help me out.

antonio-openroad, 2022-03-15:
You can change the app version to 3.6.2 and build number to 52 to make it work.
(Just make sure that the version and build numbers are the same for the 3 targets)

[image]

Poojamok, 2022-03-19:
ok got it, thank you so much.

closed 2024-02-20 by antonio-openroad