adafruit/Bluefruit_LE_Connect_v2 #61: UART mode limited to 20 characters

jerryneedell, 2024-01-17:
I am only able to send up to 20 characters via the UART Mode.
With the older version of the MacOS desktop app (version 2.2.2) This limit did not apply. I can send longer messages.

If a  message longer than 20 characters is entered, it is echoed but it does not appear to be sent.
No errors are reported.

Is the 20 character limit known and intended?

dhalbert, 2024-01-17:
Confirmed this is an issue. Tested iOS app version 3.7.0 on iOS 17.2.1 with https://github.com/adafruit/Adafruit_CircuitPython_BLE/blob/main/examples/ble_uart_echo_test.py on a Feather nRF52840 running CircuitPython 8.2.9 and also CircuitPython 7.3.3.

antonio-openroad, 2024-02-20:
There is no character limit in the app. If the message to be sent is longer than the MTU (Maximum Transfer Unit) negotiated during the BLE connection, the app will slip the text into multiple messages and send them sequentially.

I tested the iOS and macOS versions (3.7.0) and was able to send more than 20 characters. 
I used a Bluefruit LE Friend connected to a computer and CoolTerm to check the data received by UART. 

[image]

So perhaps there could be a problem with the libraries used in CircuitPython?

Adding some more information, 20 bytes was the default MTU for BLE connections for many years. Modern devices usually negotiate a bigger MTU (around 512 bytes) but is not unusual for it to set it to 20 bytes. 
So my guess is that the text is split in multiple messages, but the CircuitPython library is only processing the first one (which is 20 bytes long).

dhalbert, 2024-10-17:
This issue is really due to the default `max_length` being 20 for characteristics in the `adafruit_ble` library. But a BLE client using NUS (Nordic UART Service) normally expects the maximum writable length to be the MTU size - 3. That includes the Bluefruit LE Connect apps. So it is a library issue. I'm closing as a dupe, in favor of https://github.com/adafruit/Adafruit_CircuitPython_BLE/issues/94.

closed 2024-10-17 by dhalbert